willem-pennings/balancing-cube #1: Include parts list?

x46085, 2024-02-19:
Hello!

I've set about trying to build this, and ordered all the parts listed in the Article and Readme / in repo. 

I've ordered:
- motors
- motor controllers

I'm currently missing:
- the exact countersunk screws & count
- the capacitors by type and count
- the other transistors / parts seen on the PCBs
- the exact small radial ball bearing used

A picture of the PCB without the controllers and ESP32 in place would be helpful, especially to see the IMU placement and make sure I have it correct. 

Thank you for creating this project! It's enjoyable and inspiring to build.

willem-pennings, 2024-02-21:
I suppose you haven't found the BOM yet. It is available in the repository. I did notice that I forgot to add the ball bearings. I have added them now.

closed 2024-02-21 by willem-pennings

x46085, 2024-02-21:
Awesome, thanks!