willem-pennings/balancing-cube #2: Clarification Opportunities

x46085, 2024-03-27:
Hello! 

First off I want to reiterate how amazing your work here has been to replicate. The design work of the CAD models and electronics are truly world class; the pieces first together beautifully and things are relatively easy to follow.

Some discoveries I've had while building this:

1. Unclear on Routing Error / Bodge Wire reference: in the website you have a picture showing a mistake made on the routing for the motor controllers and how you corrected it with a bodge wire bridge. It's unclear to me if I need to replicate this, or if you corrected the routing in the final PCB design. Could you help explicitly clarify either way so I know what I should do?

2. Well-lit / High Res Picture to help with placement: I've struggled to determine if all my placed components are in the correct orientation. I found and followed the schematic in most cases, but I would find it extremely useful to have a well-lit, close-up top-down picture of the board (before placing the dev board). There is one dark picture on the website where I could see my fuses were backward and fix it, and I think the TVS Diodes have the cathode / grey stripe to the right, but I can't quite see.

3. I noticed there is no description of the steps used to flash the dev board. While I'm capable of bumbling through basic Arduino setup and the trial and error process of installing Expressif plug-ins until I find the right things to target this board, and figuring out which connection to plug into and what port this resolves to in Arduino, having any sort of steps laid out as a template / what worked before would greatly assist me in getting it right sooner and more easily.

Again this is a beautiful work of art! Thank you for building & sharing.

willem-pennings, 2024-04-01:
Hi! Thanks for the remarks. That is good to hear. Please share pictures of your build with us, either here or by sending me an e-mail through my website. That would be nice.

Regarding your questions:

1. The hardware errata have been fixed in the files that are available. There is one very minor issue which I have not fixed since it can be easily corrected in software: the pin that I had intended to use for motor speed feedback on motor driver 3 is connected to ADC2 on the ESP32. This ADC does not seem to work well when WiFi is enabled on the ESP32. However, the second analog output is connected to ADC1, hence I just modified the configuration file to swap the signals for the third driver. This works fine. The other pin is connected for motor current feedback purposes, but this signal is not needed and therefore unused.
2. I'll try to add some pictures later. For 99% of your issues, inspecting the schematic and using the 3D viewer in KiCad will answer all your questions. However, I noticed that the 3D viewer actually displays the TVS diodes incorrectly. The schematic also shows a bidirectional TVS diode whereas in reality it's just a undirectional one. I'll try to fix that too.
3. I'm assuming a certain level of competence. Arduino projects are already very accessible and I know that there are specific issues that one might run into when flashing ESP32 boards. However, I'm not going to cover them here since many tutorials are available concerning these matters.
4. The battery cable is a custom job. It uses XT60 connectors. You'll have to build it yourself or order something similar (there should be many options available). I recommend using AWG16 wire. It's about 150 mm long, but again, the required length depends on how you want to route it exactly.
5. This doesn't really matter since you can just re-assign each of the x, y and z axes in software by modifying `pins.h`. Do what works well for you. In my case, drivers 1, 2 and 3 are connected to the motors that control the y, z, and x axes, respectively.

Hope this helps. I'll keep the issue open until I add the pictures and fix diode polarity issue in the KiCad project.

x46085, 2024-04-01:
For sure! it's coming together:
[image]
and:
[image]

The second is me trying to program the motors. Did you program like this directly via this custom motherboard? or did you use Maxon's official board first? Still working through trying to get it working.

Just to clarify, I should or shouldn't have these bridges?:
[image]

willem-pennings, 2024-04-02:
Thanks! You can remove the bodge wires that you added to the motherboard. Looks like you've got the polarity of the TVS diodes right, the band should be pointing towards the corresponding fuse.

I have added some notes and tips to the main readme file which you'll want to check out!

x46085, 2024-04-02:
Indeed! I think these new comments reveal the missing piece that was preventing me from commissioning the motor controllers properly. I've got the official Maxon motherboard as well as the referenced tool ordered now. Thanks for the additions! Will post my progress (or email you if you prefer) once I get them.

Manny7115, 2024-04-24:
> Indeed! I think these new comments reveal the missing piece that was preventing me from commissioning the motor controllers properly. I've got the official Maxon motherboard as well as the [...]

Have you been able to get your cube up and running?

If so, what steps did you take to calibrate the gryo and accelerometer?

I have a built cube, but cant seem to get it to stand on any edge. Currently the cube will only enable the motor controllers if it's placed on a specific edge with the wheels towards the top and not towards the bottom like in the video.

As a side note, I had to change some of the pinouts in the pins.h based on the schematic routing. Additionally the battery board schematic is missing the GND connection from the OUT connector to the GND plane, fix this before you send the board to be produced.

willem-pennings, 2024-05-29:
> Have you been able to get your cube up and running?
> 
> If so, what steps did you take to calibrate the gryo and accelerometer?

Did you use the script that I provide? It gives some instructions on how to calibrate the gyro and accelerometer. You can verify whether your work was correct by logging the output of the AttitudeEstimator while placing the cube in certain orientations and/or rotating it about the various axes.

Manny7115, 2024-05-29:
I did use the calibration program and was able to configure the gyro but was a little unclear on how to calibrate the accelerometer or what gains to set to 0 for calibration. I believe my calibration values for the accel are correct since I am getting around 9.8 for the "Calibrated: Acc:" values on the different orientations and 1 for the G-value logging that I added.

What kind of values or ranges should I be looking for to determine if the main cube software is working as intended for the AttitudeEstimator?

Currently I have not been able to get the cube to balance on a corner, I have tried adjusting the PD gains of the main controller and of the wheels (I assume that Kpw and Kdw are the gains for the wheels). However with a little tweaking I have been able to get the edge balancing to work for a while but it seems that something is not entirely correct since the Phi value slowly creeps up and the qe1 and qe3 are increasing over time even though the wheels for those axis are not moving. This could all be related to the accelerometer or gyro since the signs of some of the gains were different than what you had.

cumhuronat, 2024-05-31:
I'm still confused about the TVS diodes. Both the schematics and the BOM list bidirectional diodes. The part number on the BOM, CDDFN2-T3.3B, is also bidirectional. If these diodes should be unidirectional, what would be the correct part number?

cumhuronat, 2024-06-01:
Also the [hardware reference of escon module](https://www.maxongroup.com/medias/sys_master/root/8834400092190/466023-ESCON-Module-24-2-Hardware-Reference-En.pdf) 24/2 suggests using CDDFN2-T3.3LC which is the predecessor of CDDFN2-T3.3B both of which are bidirectional.

cumhuronat, 2024-06-01:
Never mind, I just understood that you were talking about the diodes on the supply voltage :) Thank you so much for open sourcing this btw.

willem-pennings, 2024-06-29:
Considering the various design and documentation improvements that have been made since this issue was opened (also the contribution by Cumhur), I will now close this issue.

closed 2024-06-29 by willem-pennings