willem-pennings/balancing-cube #5: Overcurrent Error on Escon Controller

cumhuronat, 2024-06-20:
I keep getting Overcurrent Error randomly in a few seconds after the cube starts balancing and it causes the cube to fall. The error appears randomly in one of the 3 escon controllers. The error is both visible in the escon studio and with the status led (2 red blinks). I tried regulation tuning as well but it didn't help. Any ideas what might be the problem?
[image]

cumhuronat, 2024-06-20:
Thanks for sharing your pid values, will try them and let you know the result. Thanks again for making this opensource!

willem-pennings, 2024-06-20:
Thanks, I'm curious to know if this gives better results. I just noticed that the gains in my configuration files are wrong so I updated them with the values posted above. It probably explains your issue..!

cumhuronat, 2024-06-20:
It works like a charm now! Thank you
https://github.com/willem-pennings/balancing-cube/assets/529671/bb86d19a-39b8-46b5-92c9-854b028f8558

willem-pennings, 2024-06-20:
Excellent, congratulations!

closed 2024-06-20 by willem-pennings