willem-pennings/balancing-cube #8: Recent updates?

mhkeller, 2024-10-07:
I noticed that in [this comment](https://github.com/willem-pennings/balancing-cube/issues/7#issuecomment-2274198622), you mentioned that you were working on a large update with some new features – which I'm hoping includes the braking ability you mention in your original blog post that would allow it to jump up to its edge and its point. If you have a sense of your timeline in pushing this update out, please let me know. I'm scoping out starting this as a project but would prefer to wait until the new info is there. Thanks again for putting this together and for making it so accessible.

willem-pennings, 2024-10-11:
I still plan to update the controller software at some point to resolve some issues and add a few new features. I don't foresee hardware updates currently. The jumping feature is quite advanced and I haven't decided on a concept yet, so I personally wouldn't wait for it if you're keen on building this. While the cube design already has some mounting points for components related to the jump up feature, I can't guarantee that the design will be backwards compatible with the current cube design.

mhkeller, 2024-10-15:
Thanks for the update. I'm in no rush and have a few few projects still on my plate!

liy012345, 2025-05-06:
Hi，earlier I noticed that the cube could only time balance for about 10-20 minutes, and I'm wondering what causes this, zero drift in the gyroscope, or steady state error in the PD controller? Is there any way to solve the problem of the cube balancing time length, (can I theoretically turn it until the battery is dead? Or detecting how many volts the LiPo battery is below to turn off the output;)