willem-pennings/balancing-cube #10: Optional Switch for balancing modes

hexa1992, 2024-12-14:
I have developed a small circuit board that connects to the pin header of the servo motors. I would like to adjust the code so that, before powering on the cube, various settings can be configured using the switches.

In my opinion, the following settings would make sense:
	•	Selection of the balancing mode
	•	Activation of OTA update capability

Unfortunately, my programming skills are not sufficient to adapt the code accordingly.
I have already managed to successfully activate the corresponding code lines using **_if_** statements in the main program. However, this unfortunately does not work in the **_parameters.h_** file, as **_if_** statements are not possible there.

I would greatly appreciate a suggestion for a solution!
I’d be happy to upload the PCB layout here if you’re interested!
[image]
[image]
[image]
[image]

willem-pennings, 2024-12-15:
Hey,

Looks like a neat little board. You could try modifying the code as follows to make it work.

The references as defined in `parameters.h` are only used in `trajectory_attitude.cpp`. However, the orientation is set by uncommenting the desired one, so the controller only "knows" one reference at run time. In your case, you want to determine the orientation at runtime, so it makes more sense to build a generic `if` statement in `cube_controller.ino` which defines the orientation (one of three) based on a digital pin reading (more on that below). It could look something like this (pseudocode):

```
if (x edge mode) {
  qu0 = a;
  qu1 = b;
  qu2 = c;
  qu3 = d;
} else if (y edge mode) {
  qu0 = e;
  qu1 = f;
  qu2 = g;
  qu3 = h;
} else {
  // corner mode.
}

Considering that `controller_attitude.cpp` currently reads the orientation directly from the values as defined in `parameters.h`, which we do not want in your case, you could for example add a method to this class to set the reference orientation explicitly, like `set_reference()`, instead of copying the values from `parameters.h` in the constructor. You would then call this method using four arguments (the `qu` values as determined using the code above) in the `setup()` method in `cube_controller.ino`. The class is now ready for "dynamic" runtime reference selection.

To actually read the state of the DIP switches in your program, I would advise using `digitalRead()` (and `pinMode()`) in the `setup()` function in `cube_controller.ino`. Certain combinations of switches would result in certain balancing orientations, so you would put the output of the `digitalRead()` calls in the `if` block above.

If you want to change the orientation without resetting the microcontroller, you would not (only) read the DIP switches in the `setup()` function. but instead in a function which you call periodically, perhaps using a timer construction like I already do in the program to for example call the `controller()` function periodically.

I hope this helps. Do let us know whether you can get it to work.

hexa1992, 2024-12-16:
I gave the software another try. I was able to successfully modify the code lines in the `cube-controller.ino`. Unfortunately, I couldn't implement the changes involving the constants in `parameters.h`.

My code section works and has been tested by me. Depending on the switch position, you just need to adjust the corresponding block in `parameters.h`, and everything will work perfectly.

I made the following changes to `cube-controller.ino`:

I´ve added 
```
#include <set> // Für Tracking bereits ausgegebener Nachrichten

// Globale Datenstruktur zum Speichern bereits ausgegebener Nachrichten
std::set<String> loggedMessages;

``` after line 27.
This is only needed for debugging purposes in the serial monitor.

I´ve added 
```
  // Configure DIL-Switches as inputs
  pinMode(DILSW1, INPUT_PULLDOWN); // internal pulldown active

  // Mode selection
  modeselection();
``` after line 70.

I´ve added 
```
// Logging-Funktion
void logMessage(const String &functionName, const String &message, const String &values = "")
{
  String fullMessage = "[" + functionName + "] " + message;
  if (values != "")
  {
    fullMessage += ": " + values;
  }

  // Prüfen, ob Nachricht bereits ausgegeben wurde
  if (loggedMessages.find(fullMessage) == loggedMessages.end())
  {
    Serial.println(fullMessage);
    loggedMessages.insert(fullMessage); // Nachricht speichern
  }
}
```  and
```
void modeselection()
{
  if (digitalRead(DILSW1) == 1)
  {
    MODE = 2; // balancing on corner
    logMessage("modeselection", "Mode selected", "Corner balancing activated");
  }
  else
  {
    MODE = 0;
    logMessage("modeselection", "Mode selected", "X side balancing activated");
  }
}
``` after line 129, where `void logMessage` is only needed for debugging purposes in the serial monitor.

I´ve changed `att_tra.generate();` in line 185 to
```
if (MODE == 2)
    {
      att_tra.generate();
      logMessage("controller", "generate() method", "activated");
    }
    else
    {
      logMessage("controller", "generate() method", "deactivated");
    }
```

In line 210 i´ve changed 
```
// Apply torques to motors
  // When balancing on the x side, comment out motor 2 and 3. When balancing on the y side, comment out 1 and 3.
  motor1.set_torque(tau_1);
  motor2.set_torque(tau_2);
  motor3.set_torque(tau_3);
``` to
```
// Apply torques to motors
  // When balancing on the x side, comment out motor 2 and 3. When balancing on the y side, comment out 1 and 3.
  if (MODE == 0)
  {
    // Nur Motor 1 aktivieren
    motor1.set_torque(tau_1);
    logMessage("controller", "Motors activated", "Motor 1 only");
  }

  if (MODE == 1)
  {
    // Nur Motor 2 aktivieren
    motor2.set_torque(tau_2);
    logMessage("controller", "Motors activated", "Motor 2 only");
  }

  if (MODE == 2)
  {
    motor1.set_torque(tau_1);
    motor2.set_torque(tau_2);
    motor3.set_torque(tau_3);
    logMessage("controller", "Motors activated", "Motor 1, 2 & 3");
  }
``` 

also i added the following to `pins.h`:
```
// Variables for 6 DIL-Switch at Servo Inputs
const int DILSW1 = 1;
const int DILSW2 = 2;
const int DILSW3 = 42;
const int DILSW4 = 41;
const int DILSW5 = 40;
const int DILSW6 = 39;
``` 
When the switch DILSW1 is closed, MODE 2 is activated, and the cube balances on its corner. When the switch is open, MODE 0 is activated, and the cube balances on its X-axis.

If you now modify the corresponding part of the code in parameters.h, the change will work!